aerosense-ai/data-gateway #42: Update sensor_time_offset for each window

time-trader, 2022-02-20:
## Feature request

### Use Case

Sensor node internal clock can reset itself, if the sensor node crashes and rests. If such event occurs, sensor_node_offset should be recomputed. 

### Current state

Currently sensor_time_offset is computed only once per gateway session. IF the internal sensor node clock is reset during the gateway operation, and the connection is reestablished, then previously computed offset is no longer correct.

time-trader, 2022-02-20:
@WeikangKong, @tommasopolonelli since you introduce cross-node synchronization, do you have something like a "master clock"? If so, what is t=0 for that clock?

WeikangKong, 2022-02-20:
> @WeikangKong, @tommasopolonelli since you introduce cross-node synchronization, do you have something like a "master clock"? If so, what is t=0 for that clock?

t=0 would be around the moment when the base station is turned on.

cortadocodes, 2022-02-21:
Closed by #25

time-trader, 2022-02-21:
> > @WeikangKong, @tommasopolonelli since you introduce cross-node synchronization, do you have something like a "master clock"? If so, what is t=0 for that clock?
> 
> t=0 would be around the moment when the base station is turned on.

@WeikangKong  you mean when the bt receiver is plugged in, or the gateway script started?

WeikangKong, 2022-02-21:
> @WeikangKong you mean when the bt receiver is plugged in, or the gateway script started?

when the bt receiver is plugged in (power on)

time-trader, 2022-02-21:
oki thanks.

time-trader, 2022-03-28:
Sensor_time_offset was not computed correctly for recent tests.

cortadocodes, 2022-03-29:
I think this is because the offset is reset in the `Reader` process - this attribute value change won't be communicated to the `Parser` process. I'll need to make the attribute process-safe or only update the handles in the `Parser` process.

PR #49 (2022-03-30): Update handles in parser process instead of reader process

## Contents ([#49](https://github.com/aerosense-ai/data-gateway/pull/49))

### Fixes
- Update handles in the `Parser` process instead of the `Reader` process
- Closes #42
closed 2022-03-30 by time-trader