aerosense-ai/data-gateway #76: Allow json5 routines and configurations

thclark, 2022-06-28:
## Feature request

It would be useful to be able to comment the configuration files, is there any easy way of accepting JSON5, with its inbuilt comments?

Example comments in a `routine.json` file:
```
{
  "commands": [
    ["startSync", 0],
    ["selMsgDest 3", 0.1],
    ["getBattery", 0.2],
    ["startBaros", 0.3],
    ["selMsgDest 4", 1.1],
    ["getBattery", 1.2],
    ["startBaros", 1.3],
    ["selMsgDest 5", 2.1],
    ["getBattery", 2.2],
    ["startBaros", 2.3],
    ["selMsgDest 3", 60.1],
    ["stopBaros", 60.2],
    ["selMsgDest 4", 61.1],
    ["stopBaros", 61.2],
    ["selMsgDest 5", 62.1],
    ["stopBaros", 62.2],
    ["stopSync", 65],
    ["enterHighSpeed", 65.1],
    ["selDevice 3", 70], // (wait until the connection parameter is updated, you can tell that with the constatpacket number)
    ["readSens", 75], // allow 1 minute download per minute collection
    ["selDevice 4", 135], // (wait until the connection parameter is updated, you can tell that with the constatpacket number)
    ["readSens", 140],
    ["selDevice 5", 200], // (wait until the connection parameter is updated, you can tell that with the constatpacket number)
    ["readSens", 205],
    ["exitHighSpeed", 265],
    ["selMsgDest 3", 270.0], // (now we reconnect all the nodes and put them into sleep)
    ["sleep", 270.1],
    ["selMsgDest 4", 271.0],
    ["sleep", 271.1],
    ["selMsgDest 5", 272.0],
    ["sleep", 272.1],
    ["selMsgDest 3", 330.0], // (wake up nodes and allow them to re-establish connection parameters)
    ["wakeUp", 330.1],
    ["selMsgDest 4", 331.0],
    ["wakeUp", 331.1],
    ["selMsgDest 5", 332.0],
    ["wakeUp", 332.1]
  ],
  "period": 360
}
```

cortadocodes, 2022-06-29:
Would this require us using [`pyjson5`](https://pypi.org/project/json5/`)? It's apparently is up to 6000 times slower than `pyjson`.

Alternatively, we could write something (or use an existing solution) that will strip the comments before they're parsed by `pyjson`

Kijewski, 2022-07-08:
I maintain [pyjson5](https://pypi.org/project/json5/) (which is different from [json5](https://pypi.org/project/json5/)), and has the same or a better performance than the builtin json in many cases: https://pyjson5.readthedocs.io/en/latest/performance.html

cortadocodes, 2022-07-08:
Ah my mistake! Are those two links supposed to go to the same page? I drew my conclusion from the description [on this page](https://pypi.org/project/json5/)

cortadocodes, 2022-07-11:
Cool cool! I think we can look at using `pyjson5` then