aerosense-ai/data-gateway #106: Create a separate table/tables with sensor coordinates information.

time-trader, 2022-09-21:
## Feature request

### Use Case

Currently sensor coordinates are stored in the "configuration" table. The metadata for each configuration already contains ids for other data such as blade id, wind turbine id. Lets keep the configuration table static (i.e. only add row to it as new configurations are used).

Meanwhile, Information about blade, or sensor positions can change as we go along. For example we don't know the exact position of sensors when we install the sensors, but as we process photogrammetry data we can edit the position table/tables.

cortadocodes, 2022-09-21:
Thanks for raising. When do you need this completed by?

time-trader, 2022-09-21:
Not urgent. More of a conceptual note. We  are probably going to add this data manually in colab notebooks for now.

cortadocodes, 2022-09-27:
Also: have a sensor conversion constants table

cortadocodes, 2022-09-27:
@thclark what do you think about this?

time-trader, 2023-02-27:
@tommasopolonelli, rightfully noted that as we start to do more measurement campaigns and are about to install more nodes, properly storing this metadata and integrating it with aerosense-tools API became a more pressing issue.

@thclark  any input on this, or should we go ahead and do this?

cortadocodes, 2023-05-18:
To do:
- [ ] Create table with `coordinates_table.json` column names
- [ ] Store the other schemas in `strands` (sections and IMU schemas)
- [ ] Update `data-gateway` to store an ID for each sensor installation instead of the coordinates. It should do nothing else with coordinates
- [ ] Add function to `aerosense-tools` to add coordinates that also validates input coordinates against schema

closed 2023-06-07 by cortadocodes

thclark, 2023-06-07:
@cortadocodes can I just check that any legacy coordinates that are hard coded now are set up to use the table?

cortadocodes, 2023-06-07:
I don't think there are any real legacy coordinates, I think only the default values of 0 are in there from before

cortadocodes, 2023-06-07:
@time-trader am I correct?