aerosense-ai/data-gateway #107: Add sessions table

cortadocodes, 2022-10-13:
We want to add some more metadata about each session that's run on the data gateway that includes. This metadata would be constant for each session, so it doesn't seem to make sense to duplicate it on every sensor datapoint. Some of the metadata can just be set by the user, but some of it needs to be calculated automatically from sensor data.

## Requested metadata
index | start_time | end_time | baros | IMU | mics | diffP | ratio good data | average blade rotational speed | temperature | zero_val | rssi_mean?
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
int? | date | date | boolean | boolean | boolean | boolean | double | double | double | boolean | double

See more [here](https://github.com/aerosense-ai/notebooks/blob/main/pre_process.ipynb).

cortadocodes, 2022-10-13:
@thclark @time-trader what do you think about this?

cortadocodes, 2022-10-13:
Julien would like this capability in roughly the next month.

time-trader, 2022-12-12:
A comment to sum up the discussions on the topic till now:
1. We should distinguish between "measurement campaign" and "measurement session". The first one lasts for a day up to several months on a specific installation, and consists of many "measurement sessions". It is somewhat similar to the "installation" concept, but not entirely the same thing. (Unless we redefine what "installation" means).  "Measurement sessions" are continuous periods of sampling separated from each other by non-sampling stretches of time (Usually during these  non-sampling periods the data will be received from the node cache and sent to the cloud). 
1.1. Measurement Campaign data/metadata persistence is addressed by #108 
1.2. Measurement Sessions and their times can be extracted from RawSignal (aerosense-tools) measurement series using " extract_measurement_sessions" present in[ new 0.4.0 version of aerosense-tools.](https://github.com/aerosense-ai/aerosense-tools/pull/13#issue-1466892853)

closed 2023-03-09 by cortadocodes