aerosense-ai/data-gateway #116: Cloud function fails to add sensor data.

time-trader, 2023-03-10:
## Bug report

### What is the current behavior?
Cloud function will not add sensor data to greta.sensor_data because there is no column named "measurement_campaign_reference" in that table. 

### What is the expected behavior?
There is no need to have measurement_campaign_reference column at all in greta.sensor_data,  Including this column makes measurement_campaign table redundant, because it would be simply generated with a query over sensor_data.

cortadocodes, 2023-03-13:
The `measurement_campaign` column is just the reference for the measurement campaign (not the rest of the data associated with it) and was added to the `sensor_data` table so you're able to tell which data was produced in which measurement campaign.

cortadocodes, 2023-03-13:
I've just added the new columns to the tables

closed 2023-03-13 by cortadocodes