aerosense-ai/data-gateway #126: Reduce BigQuery costs

thclark, 2023-06-21:
## Feature request

We need to reduce spiralling table costs. I thought I’d look into the best strategies for sorting this beforehand, and figured that rather than barfing out suggestions in the meeting it might be better to put my thoughts down on paper.

### Background:

BigQuery is columnar, so any query for multiple column values must traverse every row in the dataset, even if you’re filtering out rows. This means that if you query for data in the last 24 hours, thousands of values from before 24 hours ago still get loaded whilst processing the query. We’d not anticipated this, and it was pointed out by Yuriy late last year so we’ve known about it for a while.

Our research back then suggested that clustering and partitioning might be good strategies for reducing this (they both have pros/cons) but we felt it wasn’t worth it at that stage, as the DB was costing on average less than a hundred bucks a month. There’s no point spending a few thousand pounds to save a few hundred. 

### Today’s research:

I’ve spent today planning and researching more deeply into clustering and partitioning. I think that:

 - We won’t totally eliminate query costs (of course) but should be able to dramatically curb them.
 - this is likely to have the corollary benefit of speeding response times on the dashboard.
 - We can do so in a way that doesn’t scale as we do more installations / measurement campaigns.
 - We should be able to achieve the same cost reduction with partitioning+clustering as we could get with clustering alone.
 - Partitioning allows improved estimation of query cost a priori, whereas with clustering you can’t know a priori to running a query how its cost is affected.
 - We don’t estimate query costs a priori for any of our features so I don’t think Partitioning has any real benefit for us.
 - Partitioning can only be done on one column, whereas clustering has the benefit of being able to specify multiple columns in priority order. So even if we did bother with partitioning, we’d also have to cluster the table to achieve the same benefit.

### I think we should:

 - Cluster the sensor_data table and possibly the microphone_data table while we’re at it. Cluster indexes should be in this order:
   - installation_reference
   - measurement_campaign_reference
   - sensor_type
   - datetime
 - But [we can’t apply clustering to the existing table(s)](https://stackoverflow.com/a/52077693/3556110) meaning we need a table migration

## Plan of Action
    - [ ] review our existing queries to check my suggested clustering strategy and index order
    - [ ] use the same schema as the existing tables, but with added clustering, to create new data tables.
    - [ ] halt all data acquisition (to prevent loss at ingress)
    - [ ] run a job to copy all data from unclustered to clustered tables
    - [ ] (meanwhile) update aerosense-tools, the data-gateway ingress and the dashboard to point to the new tables
    - [ ] deploy the updated set of tools
    - [ ] verify new tables as being duplicates
    - [ ] restart data acquisition which should ingress to new tables and verify ingress.

I think if we all do this as a group, we could block out a whole day and have it just done. It’ll be inefficient on our time doing it that way, but it’ll get it over with, avoiding asynchronous communication delays and spanning across many days.

thclark, 2023-06-21:
Note: Additional query to get count is undertaken before the `get_sensor_data` query:
```py
    count_query = f"""
        SELECT COUNT(datetime)
        FROM `{DATASET_NAME}.sensor_data_{sensor_type_reference}`
        WHERE datetime BETWEEN @start AND @finish
        AND installation_reference = @installation_reference
        AND node_id = @node_id
    """
```

thclark, 2023-06-21:
## Materialized view queries

```
SELECT datetime, sensor_value[ORDINAL(1)] as filtered_rssi, sensor_value[ORDINAL(2)] as raw_rssi, sensor_value[ORDINAL(3)] as tx_power, sensor_value[ORDINAL(4)] as allocated_heap_memory, configuration_id, installation_reference, node_id FROM `aerosense-twined.greta.sensor_data` 
WHERE sensor_type_reference='connection_statistics'
AND IS_NAN(sensor_value[ORDINAL(1)]) IS FALSE
AND IS_NAN(sensor_value[ORDINAL(2)]) IS FALSE
AND IS_NAN(sensor_value[ORDINAL(3)]) IS FALSE
AND IS_NAN(sensor_value[ORDINAL(4)]) IS FALSE;
```

cortadocodes, 2023-06-21:
Comparing querying the non-clustered (left) and clustered (right) sensor data tables - clustered comes out much better!

[image]

closed 2023-06-21 by cortadocodes

time-trader, 2023-06-21:
After looking more into it, it seems like clustering based on "datetime" has to do with performance optimisation rather than fine grain control on amount data of processed :
So for the future, if need, to optimise costs further for more scalable system we should probably use hybrid approach

**Partitioning** daily +

**Clustering** partitioned tables with this order: 
- installation_reference
- node_id
- sensor_type_reference

In this case there is no need to cluster based on measurement_campaign_reference, as these usually would last more than one day.

Alternatively, if we choose not to partition, the order can be:
- measurement_campaign_reference
- node_id
- sensor_type_reference

time-trader, 2023-06-23:
Given it more thought: each sensor type attachment we produce can have its own id like "kraken-of-the-abyss-barometers", "bloom-of-the-summer-barometers", and "submarine-of-the-zürichsee-differential-pressure" etc., with those ids included in the "installation" table. The sensor data_table then can entirely ditch 3 columns: "installation_reference", "node_id" and "sensor_type"  and could be clustered just by these unique ids. This plus daily partitioning will be fully scalable.